pcengines/coreboot #456: details of i2c support on APU2

sniku, 2020-12-26:
First of all, thank you for continuing to work on the APUx platform and providing new BIOS updates regularly! 

My goal is to get an OLED LCD screen to work on APU2 on FreeBSD. 

I have spent a considerable amount of time trying to get i2c to work on *BSD without success. I somewhat succeeded on Linux, but only partially.  I'll lay out everything I've learned and how far I've come and then, at the end, ask for help.

Hopefully, this post will be useful for others. i2c information for APU2 is sparse on the internet.

## Hardware information
APU2/3/4 SOC has `AMD A55E Hudson-1 FCH`, which provides the `PIIX4` device. 
APUx motherboards have pins for i2c exposed, but the header isn't soldered. 

I have a small 128x32 ssd1306 OLED display.    

## i2c support on Linux

Dmesg prints these three lines
```
[    7.382308] piix4_smbus 0000:00:14.0: SMBus Host Controller at 0xb00, revision 0
[    7.382334] piix4_smbus 0000:00:14.0: Using register 0x02 for SMBus port selection
[    7.383029] piix4_smbus 0000:00:14.0: Auxiliary SMBus Host Controller at 0xb20
```

On Linux, there's a kernel module that implements i2c for piix4. It's called `i2c-piix4`. Once this module is loaded, i2c can be listed with `i2cdetect -l`. On APU2 four Bus adapters are listed. 

```
root@debapu:~# i2cdetect -l
i2c-3   smbus           SMBus PIIX4 adapter port 4 at 0b00      SMBus adapter
i2c-1   smbus           SMBus PIIX4 adapter port 2 at 0b00      SMBus adapter
i2c-4   smbus           SMBus PIIX4 adapter port 1 at 0b20      SMBus adapter
i2c-2   smbus           SMBus PIIX4 adapter port 3 at 0b00      SMBus adapter
i2c-0   smbus           SMBus PIIX4 adapter port 0 at 0b00      SMBus adapter
```

Note, all found devices are "SMBus adapter", not "I2C adapter" as I was expecting. 

To detect what functionality the devices support, the following command is executed:
```
root@debapu:~# i2cdetect -F 0
Functionalities implemented by /dev/i2c-0:
**I2C                              no**
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               no
SMBus Block Write                yes
SMBus Block Read                 yes
SMBus Block Process Call         no
SMBus PEC                        no
**I2C Block Write                  no**
**I2C Block Read                   no**
```

i2cdetect says that those are not i2c adapters. Most/all i2c LCDs require `I2C Block Write` functionality to work. 
I'm not sure if `SMBus Block Write` functionality can be used to drive i2c LCD.  If it can, I would love to hear how to use it.

Interestingly, `i2cset` commands do work. These commands can be used to set individual pixels. A sample script driving the LCD, pixel by pixel, is pasted below.

## i2c OLED demo on Linux 

Here's a short video demonstrating that __some__ i2c functionality actually works on Linux. 

https://user-images.githubusercontent.com/442646/103160556-145b4a80-47d7-11eb-9bbf-63c75cc7b104.mp4

Here's the script used for the video.
```
[... 151 lines cut ...]
pin 08: 0       GPIO10<IN,OD>
pin 09: 0       GPIO11<IN,OD>
pin 10: 0       GPIO12<IN,OD>
pin 11: 0       GPIO13<IN,OD>
pin 12: 0       GPIO14<IN,OD>
pin 13: 0       GPIO15<IN,OD>
pin 14: 0       GPIO16<IN,OD>
pin 15: 0       GPIO17<IN,OD>
```
But I'm not sure where to go from here. 

**Serial connection through the internal COM2 port**
This is something that has been suggested on one of the numerous forum posts I've read through. I have not tried it. 
Is it feasible?

**SMBus**
Is it possible to somehow use the SMBus functionality to drive LCD?

**SPI**
I'm only listing SPI for completeness. After reading the hardware schematics, I see that the SPI header on APU is only useful for BIOS recovery. Or am I wrong?

**External I2C adapter**
Is there a hardware i2c adapter that can be connected to mPCIe or internal USB? If there's one, I would be happy to buy it and not worry about the other methods. :-)

Any comments, corrections, or general help would be much appreciated :-)

Strykar, 2021-01-02:
What did support@pcengines.ch say? You should post their response here to add to this nice i2c writeup on the APU.
I have not tried playing with i2c on the APU2 myself sadly, but I expect PC Engines support to have the answers you seek.

miczyg1, 2021-01-04:
@sniku I recall an attempt to connect LCD display to I2C/SMBus pins on APU. If SMBus write byte is compatible with I2C write byte, this is the best possible way to drive it, although it would be better to have a system driver or software written in C )or something else) to avoid going through all system wrappers around the i2c tools.

As for other solutions, you could for example use the [FTDI FT232H](https://www.ftdichip.com/Products/ICs/FT232H.htm) and connect it to an internal USB 2.0 port on the board which would enable you a pure I2C interface. FT232H should have a decent support (FreeBSD should have libftdi or something), maybe it could help. Although that also means writing some code to handle the display (in python for example).

Another solution is to check for the display's model-specific driver if it exists anywhere for Linux or FreeBSD.