pcengines/coreboot #457: CPU clock stuck at low frequency (APU2C4,

thowe-switzerland, 2021-01-18:
Basically my APU2C4 still seems to suffer from this (closed) problem: https://github.com/pcengines/coreboot/issues/196

After a reboot, the CPU works for some time (a few hours) normally and also goes up to the nearly 1000MHZ. But after some time the maximum frequency is stuck at 600MHz again until the next reboot.

**Turbostat**

> pkg add http://pkg0.isc.freebsd.org/FreeBSD:11:amd64/latest/All/turbostat-4.17_2.txz 
> rehash 
> kldload cpuctl 
> turbostat --interval 3

> turbostat version 17.06.23 - Len Brown <lenb@kernel.org>
> CPUID(0): AuthenticAMD 13 CPUID levels; family:model:stepping 0xf:30:1 (15:48:1)
> CPUID(1): SSE3 MONITOR - - - TSC MSR - -
> CPUID(6): APERF, No-TURBO, No-DTS, No-PTM, No-HWP, No-HWPnotify, No-HWPwindow, No-HWPepp, No-HWPpkg, No-EPB
> CPUID(7): No-SGX
> NSFOD /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver
> Core    CPU     Avg_MHz Busy%   Bzy_MHz TSC_MHz IRQ
>  \-      -        208     34.75   599     998     815
> 0       0       219     36.55   598     997     85
> 1       1       207     34.52   599     998     581
> 2       2       203     33.91   599     998     77
> 3       3       204     34.03   599     998     72

**Stress test also results in low bogo ops:**

> \# stress-ng --cpu 4 --cpu-method matrixprod --timeout 30 --metrics
> stress-ng: info:  [55320] dispatching hogs: 4 cpu
> stress-ng: info:  [55320] successful run completed in 30.09s
> stress-ng: info:  [55320] stressor       bogo ops real time  usr time  sys time   bogo ops/s   bogo ops/s
> stress-ng: info:  [55320]                           (secs)    (secs)    (secs)   (real time) (usr+sys time)
> stress-ng: info:  [55320] cpu                 988     30.06     75.13      0.05        32.87        13.14

Hardware: APU2C4
Verified and affected BIOS versions (basically all versions I tried):
- v4.10.0.0
- v4.11.0.6
- v4.13.0.2 

More aspects:
- OPNsense 20.7.7_1-amd64
- FreeBSD 12.1-RELEASE-p11-HBSD
- after flashrom I always do a cold start (30s powerless)
- BIOS settings default (e.g. CPU Boost enabled)

Ideally, the problem can be solved with a BIOS fix. (My hope)
Workaround: Are there ways to prevent the CPU from going down to the 600MHz? Preferably without having to turn off the Power Boost?
Or do I have to switch the APU board?

miczyg1, 2021-01-18:
@krystian-hebel any insights about it?

krystian-hebel, 2021-01-18:
@thowe-switzerland 
1) Is this APU in official case? I assume it is, but asking just to confirm.

2) Does this bug occur with CPU Boost disabled? Are there any other unexpected hangs, problems with reboot etc? We had some reports of CPB not working reliably on some platforms, while being perfectly fine on the others. This is why we introduced an option to turn it off, for a couple of releases we had it fixed as enabled. This most likely is a hardware issue for a specific unit, and not a model as a whole. Even when talking about straight overclocking in PC CPUs you can find "better" and "worse" units of the same CPU model, you just have to get lucky. Unfortunately, all of our platforms in the lab do not have these issues so we are forced to rely on users' reports only...

3) Do you have any unusual peripheral devices, storage, multiple WiFi adapters?

4) Have you tested if a non-BSD OS behaves the same way?

thowe-switzerland, 2021-01-18:
@krystian-hebel 

Thanks for having a look at it.

1. It is in the original case by PC Engines (a red one ;-) )

2. I did not test this. Maybe I will give it a try.

3. No special hardware. Only the standard 16GB SSD original by PC Engines.

4. I did not test with another OS as the hardware is a firewall in production.

I still believe in the APU. So today I ordered a new APU2E4 system (with blue case and 30GB SSD) from PC Engines. As soon as this is operational and replaces the APU2C4 I can do tests (CPB disabled, other OS).

thowe-switzerland, 2021-01-20:
New APU2E4 arrived today, is assembled, configuration is transferred, firewall is already in production. I will have an eye on that one regarding low frequency. So far operating normal and snappy.

This also means, that the affected older APU2C4 is currently a spare device and ready to execute some tests as requested by @krystian-hebel. I will get back with answers to the questions as soon as I have them.

thowe-switzerland, 2021-01-21:
Observations so far:

- The replacement APU2E4 runs with the very same BIOS (v4.13.0.2) + CPU Boost enabled + OPNsense configuration and is showing the same effect: after some hours the max Bzy_MHz remains at 599. So there is no difference between the two boards.
- The old APU2C4 (standalone test operation without plugged in NICs) in the meantime has been running fine under stress-ng but - without showing the issue, CPU goes up to 1 GHz.

So the effect seems not to be related to that one specific APU2C4 board but a general issue - at least in this specific software/load configuration.

I am still trying to provoke the issue at the old (now isolated) APU2C4 in order to understand what circumstances have to be met in order to get into this 600MHz state. Without NICs active (no IRQ on the CPUs) I could not provoke it so far.

thowe-switzerland, 2021-01-21:
Current Turbostat of the new APU2C4 (v4.13.0.2, CPU Boost enabled, OPNsense):

root@router:/home/weth # turbostat --interval 3
turbostat version 17.06.23 - Len Brown <lenb@kernel.org>
CPUID(0): AuthenticAMD 13 CPUID levels; family:model:stepping 0xf:30:1 (15:48:1)
CPUID(1): SSE3 MONITOR - - - TSC MSR - -
CPUID(6): APERF, No-TURBO, No-DTS, No-PTM, No-HWP, No-HWPnotify, No-HWPwindow, No-HWPepp, No-HWPpkg, No-EPB
CPUID(7): No-SGX
NSFOD /sys/devices/system/cpu/cpu3/cpufreq/scaling_driver
Core    CPU     Avg_MHz Busy%   Bzy_MHz TSC_MHz IRQ
\-       -       325     54.32   599     998     354
0       0       225     37.59   599     998     95
1       1       331     55.26   599     998     114
2       2       418     69.77   599     998     88
3       3       327     54.67   599     998     57
Core    CPU     Avg_MHz Busy%   Bzy_MHz TSC_MHz IRQ
\-       -       455     75.99   599     998     322
0       0       406     67.87   599     998     98
1       1       506     84.42   599     998     118
2       2       496     82.79   599     998     71
3       3       413     68.89   599     998     35

thowe-switzerland, 2021-01-23:
Interim Report

- The "CPU stuck at 600MHz" issue happens to the old APU2C4 and the new APU2E4. So the issue is not related to the one board.
- A board in my setup is only affected when it is operated as the productive OPNsense firewall under load. Within a few minutes to hours it is stuck at 600MHz after a reboot.
- A board in isolation running the same OPNsense without network traffic never showed the stuck CPU issue. Even when stressed with stress-ng.
- I even switched the Power Supply Units in order to to exclude a fault in the power supply.
- I have tested different BIOS versions without a difference.

My plan is to provoke the issue on the spare system because it is not well received by users if I constantly experiment on the production firewall. I'll try to put a simulated network load on it.

krystian-hebel, 2021-01-25:
@thowe-switzerland thanks for reports. The fact that the issue doesn't happen without NIC activity may explain why we weren't able to reproduce this on our side. I'll ask our validation team if we can do something about it, but I can't promise we will be able to start testing before weekend, platforms will be needed for testing new release soon.

thowe-switzerland, 2021-01-30:
I tested with Core Performance Boost disabled. And it seems to make no difference: Initially CPU went up to 1GHz. After some hours it was stuck at 60üMHz. I did not expect this. So I re-checked BIOS settings, and yes CPU Boost really is disabled.

thowe-switzerland, 2021-02-01:
In the OPNsense forum we try to get some more light into the darkness. In the process, we have discovered:

1. Other users are also affected.
2. The problem does not seem to occur when the powerd of FreeBSD is either completely switched off or running on the profile "Maximum".

miczyg1, 2021-02-01:
@krystian-hebel 
Ad.2. doesn't that mean the CPU driver/frequency scaler in FreeBSD is faulty? The profile Maximum should be equivalent on running the highest P-state, so no switching to lower P-states (and thus lower frequencies).

thowe-switzerland, 2021-02-01:
Is not excluded from my point of view.

What irritates me a bit, however, is the fact that it sometimes runs flawlessly for a few hours. And in the lab without NIC connected, the problem does not occur at all. Although I create strongly fluctuating load with stressng.

krystian-hebel, 2021-02-01:
@miczyg1 depends how the driver defines "maximum". If it were aware of CPB then throttling down other cores may result in higher performance when there is only one CPU-heavy process. I think we may assume that it behaves as you described, with no software P-state switching whatsoever.

Now we have to think what may be the common denominator of powerd and NIC. Perhaps an interrupt in a bad place in powerd code messes something up, but this is only a guess at this point.

miczyg1, 2021-06-24:
@thowe-switzerland with the recent v4.14.0.1 we have fixed some issues related to CPU boost and C-states which may help with the problem of idling CPUs and stuck frequencies. It should also improve the stability of the BSD systems. Please try to install the latest (v4.14.0.1) firmware and let me know if the problem persists.