pcengines/coreboot #464: Boot screen issue with OPNsense

fire-burner, 2021-02-28:
I have an APU4D4 for almost a year now.
I started with Coreboot version 4.11.0.2 and I am now running 4.13.0.2.
As an OS, Irun OPNsense on it. (issue exists for at least 20.1 20.7 and 21.1)

When booting the OPNsense installer, the OPNsense boot screen is properly shown.
Once it is installed on the SSD, the boot screen is distorted and it makes it impossible to select something on that boot screen. Quickly afterwards everything is normally displayed in the boot process:

H ____________ __________ __________

H// ____ ___ ||// ____ ||
H|| || || || ||// || || || ||__ ______ __ ____ ______ _____
H|| || || || //|| || || // ||// __ \ '' \// ||// __
H|| |||| || || || || || \ \ // || || \ \ /
H||//|||| |||| //||//\|||| ||||//\|

H++===================H===================H++=======================================@@@@@@@@@@@@@@@@@@@@@@@
H|| H|| H@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
H||HH..HBBoooott MMuullttii UUsseerr [[EEnntteerr]]@@@ @@@@@@@@@@
H||HH..HBBoooott [[SS]]iinnggllee UUsseerr| H @@@@@@@@@@ @@@@@@@@@@
13;HH..H[[EEsscc]]aappee ttoo llooaaddeerr pprroommpptt@@@@@@@@@ @@@@@@@@@@@@@@@@@@@@@@
H||HH..HRReebboooott H|| H \\\\\ //////////
H|| H|| H)))))))))))))))))))))))) ((((((((((((((((((((((
H||HOOppttiioonnss:: H|| H ////////// \\\\\
H||HH..H[[KK]]eerrnneell:: kkeerrnneell ((11 ooff 22))@@@@@@@@@@@@@ @@@@@@@@@@@@@@@@@@@@@@
H||HH..HCCoonnffiigguurO]]ppttiioonnss......| H @@@@@@@@@@ @@@@@@@@@@
H|| H|| H@@@@@@@@@@ @@@@@@@@@@
H|| H|| H@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
H|| H|| H@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
H++=======================================H++==========================
H aarryy LLiioonn''''

/bboooo]] /kssiizzee==00xx11000000 aattaa==00xx11ee33442ssyymmss==[[00xx88++00xx8188ee99ee88||
8++00xx11aaccee9922//
/bboooott//kkeerrnneell//ccaarrpp..kkoo ssiizzee 00xx1100225500 aatt 00xx226688ff000000

/bboooott//kkeerrnneell//iiff__bbrriiddggee..kkoo ssiizzee 00xxff999988 aatt 00xx2266aa00000000

looaaddiinngg rreeqquuiirreedd mmoodduullee ''bbrriiddggeessttpp''
ssiizzee 00xx558822ff00 aat0 00xx2266ee77000000 0000
/bboooott//kkeerrnneell//bbrriiddggeessttpp..kkoo ssiizzee 00xx7722aa88 aatt 00xx2266bb00000000
Boooottiinngg...... ell//ppfflloogg..kkoo ssiizzee 00xx22aaff88 aatt 00xx22774400000000
/bboooott//kkeerrnneell//iiff__eenncc..kkoo ssiizzee 00xx33ee7788 aatt 00xx2266bb88000000
KDB: debugger backends: ddbpffssyynncc..kkoo ssiizzee 00xxeebbdd00 aatt 00xx22774433000000
KDB: current backend: ddb/iiff__ggrree..kkoo //

I already had opened a case in the OPNsense community, but they assumed it must be an issue in Coreboot (https://github.com/opnsense/core/issues/4231)

[image]

miczyg1, 2021-03-01:
The SeaBIOS has legacy serial console redirection enabled, that is why you see the double-printed menu. Setting the `/boot.config` to `-S115200 -h` should help.

Here we have an installerconfig for BSD systems: https://github.com/pcengines/apu2-documentation/blob/0d29f8cb072ab6eb566716451e3a5c5a2979d27d/scripts/installerconfig
At line 186 and later we set the serial console correctly.

fire-burner, 2021-03-01:
OPNsense for some reason doesn't keep the changes in /boot.config when rebooting.
But I found something else in the OPNsense config.
VGA console was set as default console.
So I set serial as default and disabled the secondary console.
Now only the first boot line is distorted, but the menu is shown correctly :)
So I its not an issue with the Seabios. Thanks.

closed 2021-03-01 by fire-burner

miczyg1, 2021-03-02:
@fire-burner regarding the boot.config we do a little trick to make it work:

```
echo "-S115200 -h" > /boot/config;
rm /boot.config;
ln /boot/config /boot.config;
```