pcengines/coreboot #469: 2 x APU2E4 unstable with CPB enabled.

MyGithubUser01, 2021-05-09:
Hi all,

I'm having some serious stability issues with APU2E4 and CPB with BIOS 4.13.0.1 and 4.13.0.5
This is brand new hardware which was believed to be "DOA" but the replacement I got had the exact same issue.
After disabling CPB the system appears to be stable and has an uptime of a record high 4 days.

Operating system tested OPNsense 21.1 and 21.1.5.
I've tried booting from msata, sd card and USB but it gives me the same issue.
CPU Temperature is typically 54-56c and the system isn't even connected to any network just the console cable.

The system has been very unstable and is core dumping every 4-12 hours. BIOS 4.13.0.1, but did see similar issues when testing 4.13.0.5.

From the logs/console I see the following:

FreeBSD/amd64 (OPNsense.localdomain) (ttyu0)
login: MCA: Bank 1, Status 0x9400000000000151
MCA: Global Cap 0x0000000000000106, Status 0x0000000000000000
MCA: Vendor "AuthenticAMD", ID 0x730f01, APIC ID 0
MCA: CPU 0 COR ICACHE L1 IRD error
MCA: Address 0x282060
[HBSD SEGVGUARD] [/usr/local/bin/python3 (5880)] Suspension expired.
 -> pid: 5880 ppid: 1302 p_pax: 0xa50<SEGVGUARD,ASLR,NOSHLIBRANDOM,NODISALLOWMAP32BIT>

And:
"root@OPNsense:/var/db/rrd # MCA: Bank 1, Status 0xd400000000000151
MCA: Global Cap 0x0000000000000106, Status 0x0000000000000000
MCA: Vendor "AuthenticAMD", ID 0x730f01, APIC ID 0
MCA: CPU 0 COR OVER ICACHE L1 IRD error
MCA: Address 0xffff80d1ff60"

Any idea, broken hardware or bios bug?

miczyg1, 2021-05-09:
The CPB is additionally enabling core/package C6 states. I have recently discovered some bugs in coreboot around the C6 and its save state area in DRAM. It may be causing problems when CPB is enabled. The patches to coreboot are already sent, so I can test if those resolve your issue. If I understood correctly there is no need for stressing the firewall device to trigger this?

miczyg1, 2021-06-24:
@MyGithubUser01 I have left OPNsense 21.1 installer running on apu2 over a night (20 hours elapsed since i left the machine idling) with CPB enabled. Not a single MCA error on the serial console with apu2 v4.14.0.1. Could you please give v4.14.0.1 a try?

MyGithubUser01, 2021-07-02:
Hi,

Thank you very much for the update, I've now updated to 4.14.0.1 and made sure CPB is enabled (looks like it was enabled after flashing). I started the firewall about 24h ago with Serial console and WAN connected, but this morning I only had 5h of uptime and found the below in the console/log. This means it happened after 16-20h.

Looks like it doesn't happen as often/frequent as before - but I'm not sure.

WARNING: attempt to domain_add(netgraph) after domainfinalize()
pid 26616 (python3.7), jid 0, uid 0: exited on signal 11 (core dumped)
[HBSD SEGVGUARD] [/usr/local/bin/python3 (88561)] Suspension expired.
 -> pid: 88561 ppid: 82917 p_pax: 0xa50<SEGVGUARD,ASLR,NOSHLIBRANDOM,NODISALLOWMAP32BIT>
MCA: Bank 1, Status 0x9400000000000151
MCA: Global Cap 0x0000000000000106, Status 0x0000000000000000
MCA: Vendor "AuthenticAMD", ID 0x730f01, APIC ID 0
MCA: CPU 0 COR ICACHE L1 IRD error
MCA: Address 0xffff811f7c40
pid 95751 (python3.7), jid 0, uid 0: exited on signal 10 (core dumped)
[HBSD SEGVGUARD] [/usr/local/bin/python3 (74746)] Suspension expired.
 -> pid: 74746 ppid: 65084 p_pax: 0xa50<SEGVGUARD,ASLR,NOSHLIBRANDOM,NODISALLOWMAP32BIT>
MCA: Bank 1, Status 0x9400000000000151
MCA: Global Cap 0x0000000000000106, Status 0x0000000000000000
MCA: Vendor "AuthenticAMD", ID 0x730f01, APIC ID 0
MCA: CPU 0 COR ICACHE L1 IRD error
MCA: Address 0x63b4b76aae0

miczyg1, 2021-07-02:
Somehow I cannot reproduce it and we never run into MCA erros before.

The warning `WARNING: attempt to domain_add(netgraph) after domainfinalize()` looks suspicious. I have found a similar issue here: https://forum.opnsense.org/index.php?topic=17417.0
Maybe following this thread could help you a bit?

v1k4, 2021-07-05:
Brand new APU4D4 here. Crashing multiple times a day when CPB enabled. FW versions v4.14.0.1 and v4.14.0.2.

Currently running Proxmox on Buster and I get many this kind of errors before APU will eventually end up in hanging/crashing.

mce: [Hardware Error]: Machine check events logged
[Hardware Error]: Corrected error, no action required.
[Hardware Error]: CPU:1 (16:30:1) MC1_STATUS[-|CE|-|AddrV|-|-|-]: 0x9400000000000151
[Hardware Error]: Error Addr: 0x0000ffff86b0b9e0
[Hardware Error]: MC1 Error: Data/tag array parity error for a tag hit.
[Hardware Error]: cache level: L1, tx: INSN, mem-tx: IRD

Disabling CPB seems to make it stable for now.

MyGithubUser01, 2021-07-07:
This sounds very similar to what I'm experiencing with CPB enabled, thanks for chiming in.
The same address is mentioned: 0x9400000000000151

This also seems to be the same issue:
https://forum.netgate.com/topic/156761/page-fault-while-in-kernel-mode-on-apu2-after-bios-coreboot-upgrade/4

miczyg1, 2021-07-08:
`0x9400000000000151` is not the address but the actual 64bit MC1_STATUS register content. It simply means that the same error was triggered, but the address still may be different.  The address where it was triggered is present in this line `[Hardware Error]: Error Addr: 0x0000ffff86b0b9e0`, the error code is decoded in the following lines.

Still what is written in the forum is not exactly true. CPU Boost does not raise the memory clock frequency, it can't do that because it would require retraining the memory to the new frequency (only BIOS can train the memory, once it is done, the memory frequency is fixed).

CPB is not an overclocking feature! It simply raises the CPU clock frequency to the limits allowed by the CPU specification. Overclocking would be to go higher than what CPB provides (i.e. higher than 1400MHz).

toredash, 2021-09-26:
Wanted to chime in this. I have had the same stability issues described here with my APU2E4 for many years. It would work sometimes for weeks at a time, then it would have stability issues daily over a period of several weeks, then back to a few weeks between each failure.

I have disable CBP now, and so far it looks good. But it is only been a week, so I will have to wait a few months to really be sure.

I've found others that points to the same thing, that CBP causes issues:

https://forum.netgate.com/topic/156761/page-fault-while-in-kernel-mode-on-apu2-after-bios-coreboot-upgrade/38
https://www.reddit.com/r/homelab/comments/lokgyg/solution_to_pc_engines_apu2e4_having_constant/