pcengines/coreboot #526: How to set bios Release date?

Nemirtingas, 2025-01-14:
@pietrushnic 
Hi, I've been recompiling BIOS for our APU2 and APU3 recently (cause we need to tweak some settings). After running the whole new process of building the firmware image, I noticed that our BIOS is lacking some informations like Release Date, BIOS Revision, Firmware Revision. We don't need those entries, but I'm a bit curious on how you can set them.

```
BIOS Information
	Vendor: coreboot
	Version: v24.05.00.01_company_test
	Release Date: 01/01/1970
	ROM Size: 8 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 0.0
	Firmware Revision: 0.0
```

miczyg1, 2025-01-14:
@Nemirtingas the builds are using TIMELESS feature, the date is fixed to the Unix epoch and cannot be changed for the sake of build reproducibility.

Nemirtingas, 2025-01-14:
Alright, thanks!

closed 2025-01-14 by Nemirtingas